grblHAL/Plugin_keypad #2: Some compile errors here

stevempotter, 2022-01-13:
I have downloaded this Plugin_keypad repo and put its files in the src/keypad folder of the grblHAL_Teensy4 folder. And turned on the keypad plugin in my_machine.h, option 1: I2C. Upon attempting compiling in VScode (Windows 10), I get the following warnings and errors. Any advice on how to fix?

```
Compiling .pio\build\teensy41\src\networking\TCPStream.c.o
src\keypad\keypad.c:77:14: error: unknown type name 'setting_descr_t'
 static const setting_descr_t keypad_settings_descr[] = {
              ^
src\keypad\keypad.c:78:5: warning: braces around scalar initializer
     { Setting_JogStepSpeed, "Step jogging speed in millimeters per minute." },
     ^
src\keypad\keypad.c:78:5: note: (near initialization for 'keypad_settings_descr[0]')
src\keypad\keypad.c:78:29: warning: excess elements in scalar initializer
     { Setting_JogStepSpeed, "Step jogging speed in millimeters per minute." },
                             ^
src\keypad\keypad.c:78:29: note: (near initialization for 'keypad_settings_descr[0]')
src\keypad\keypad.c:79:5: warning: braces around scalar initializer
     { Setting_JogSlowSpeed, "Slow jogging speed in millimeters per minute." },
     ^
src\keypad\keypad.c:79:5: note: (near initialization for 'keypad_settings_descr[1]')
src\keypad\keypad.c:79:29: warning: excess elements in scalar initializer
     { Setting_JogSlowSpeed, "Slow jogging speed in millimeters per minute." },
                             ^
src\keypad\keypad.c:79:29: note: (near initialization for 'keypad_settings_descr[1]')
src\keypad\keypad.c:80:5: warning: braces around scalar initializer
     { Setting_JogFastSpeed, "Fast jogging speed in millimeters per minute." },
     ^
src\keypad\keypad.c:80:5: note: (near initialization for 'keypad_settings_descr[2]')
src\keypad\keypad.c:80:29: warning: excess elements in scalar initializer
     { Setting_JogFastSpeed, "Fast jogging speed in millimeters per minute." },
                             ^
src\keypad\keypad.c:80:29: note: (near initialization for 'keypad_settings_descr[2]')
src\keypad\keypad.c:81:5: warning: braces around scalar initializer
     { Setting_JogStepDistance, "Jog distance for single step jogging." },
     ^
src\keypad\keypad.c:81:5: note: (near initialization for 'keypad_settings_descr[3]')
src\keypad\keypad.c:81:32: warning: excess elements in scalar initializer
     { Setting_JogStepDistance, "Jog distance for single step jogging." },
                                ^
src\keypad\keypad.c:81:32: note: (near initialization for 'keypad_settings_descr[3]')
src\keypad\keypad.c:82:5: warning: braces around scalar initializer
     { Setting_JogSlowDistance, "Jog distance before automatic stop." },
     ^
src\keypad\keypad.c:82:5: note: (near initialization for 'keypad_settings_descr[4]')
src\keypad\keypad.c:82:32: warning: excess elements in scalar initializer
     { Setting_JogSlowDistance, "Jog distance before automatic stop." },
                                ^
src\keypad\keypad.c:82:32: note: (near initialization for 'keypad_settings_descr[4]')
src\keypad\keypad.c:83:5: warning: braces around scalar initializer
     { Setting_JogFastDistance, "Jog distance before automatic stop." }
     ^
src\keypad\keypad.c:83:5: note: (near initialization for 'keypad_settings_descr[5]')
src\keypad\keypad.c:83:32: warning: excess elements in scalar initializer
     { Setting_JogFastDistance, "Jog distance before automatic stop." }
                                ^
src\keypad\keypad.c:83:32: note: (near initialization for 'keypad_settings_descr[5]')
src\keypad\keypad.c:115:5: error: unknown field 'descriptions' specified in initializer
     .descriptions = keypad_settings_descr,
     ^
src\keypad\keypad.c:115:21: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
     .descriptions = keypad_settings_descr,
                     ^
src\keypad\keypad.c:115:21: note: (near initialization for 'setting_details.settings')
src\keypad\keypad.c:116:5: error: unknown field 'n_descriptions' specified in initializer
     .n_descriptions = sizeof(keypad_settings_descr) / sizeof(setting_descr_t),
     ^
src\keypad\keypad.c:116:62: error: 'setting_descr_t' undeclared here (not in a function)
     .n_descriptions = sizeof(keypad_settings_descr) / sizeof(setting_descr_t),
                                                              ^
src\keypad\keypad.c: In function 'keypad_process_keypress':
src\keypad\keypad.c:181:21: error: 'grbl_t {aka struct <anonymous>}' has no member named 'enqueue_realtime_command'
                 grbl.enqueue_realtime_command(CMD_FEED_HOLD);
                     ^
compilation terminated due to -fmax-errors=5.
*** [.pio\build\teensy41\src\keypad\keypad.c.o] Error 1
```

terjeio, 2022-01-13:
> Any advice on how to fix?

My guess this is due to the rest of the code you have is out of date, you have to [pull the driver and the submodules](https://github.com/grblHAL/core/wiki/Compiling-GrblHAL#using-git-command-line). There is no need to download the keypad plugin separately as it is already included in the submodules for the iMXRT1062 driver.

stevempotter, 2022-01-13:
Yes, that was the problem. It compiles now, thank you. Perhaps you could include a warning/advice in the keypad wiki: Get this plugin working by refreshing your whole grblHAL codebase, not just that one plugin. This might help avoid this problem for others. 

Perhaps for a new thread:
What is the status of the MPG mode, and does that require the Encoder plugin? When I tried turning that on, I got a lot of compile errors. Can I have the keypad one-char commands coming in the I2C port, and the encoder pulses coming in the UART header?

terjeio, 2022-01-13:
> . Perhaps you could include a warning/advice in the keypad wiki: Get this plugin working by refreshing your whole grblHAL codebase, not just that one plugin. This might help avoid this problem for others.

Yes, I'll do that at some point - there is a lot of documentation that needs to be written...

> What is the status of the MPG mode, and does that require the Encoder plugin?

It should be working now, and no it does not require the encoder plugin.

>  When I tried turning that on, I got a lot of compile errors. 

If you have enabled plugin/features that want to use the same resources then that may happen. Eg. there is only one UART port, if enabling the VFD spindle then MPG mode cannot be enabled at the same time. Post some of the errors and _my_machine.h_ and I can check if my tests are not good enough or if there is a real bug.

> Can I have the keypad one-char commands coming in the I2C port, and the encoder pulses coming in the UART header?

Yes, but I am not sure what you mean by having encoder pulses coming in the UART header, have you remapped those pins for encoder use? If so that may explain why enabling MPG mode causes compiler errors, it needs the UART port.

closed 2023-01-29 by terjeio