grblHAL/Plugin_keypad #7: Is there a way for me to switch between USB and MPG UART without having to reflash the microcontroller (STM32F407)?

terjeio, 2023-09-17:
> > Now here's my problem - is there a way for me to switch between USB and MPG UART without having to reflash the microcontroller (STM32F407)? Or can I switch between the USB serial and the dedicated UART serial?
> 
> See the Wiki [here](https://github.com/grblHAL/core/wiki/MPG-and-DRO-interfaces#mpg--dro-mode).

I tried different configurations of the firmware by changing the values of KEYPAD_ENABLE and MPG_ENABLE while trying to keep the USB as the main serial communication method, but I found out that if MPG_ENABLE is not defined and KEYPAD_ENABLE is - the USB works and the app on my computer (LaserGRBL) can connect to my microcontroller, but if MPG_ENABLE is defined, a connection through USB cannot be made with a PC. Why is that? Is it even possible to have a a machine, which receives g-code from USB, but can also be manually controlled with jogging commands from a "pendant"/control panel, without changing the code of the firmware? Can this problem be solved if MPG_ENABLE is not defined and only KEYPAD_ENABLE is and the jogging commands can be sent to the dedicated keypad plugin UART/I2C peripheral?

_Originally posted by @AvatarBg111 in https://github.com/grblHAL/Plugin_keypad/issues/3#issuecomment-1722419951_

terjeio, 2023-09-17:
> if MPG_ENABLE is defined, a connection through USB cannot be made with a PC.

Which board map are you using? Asking since it could be due to pin assignment clashes. I do not lose USB connection when testing with my dev board.

> Is it even possible to have a a machine, which receives g-code from USB, but can also be manually controlled with jogging commands from a "pendant"/control panel, without changing the code of the firmware?

Yes.

> Can this problem be solved if MPG_ENABLE is not defined and only KEYPAD_ENABLE is and the jogging commands can be sent to the dedicated keypad plugin UART/I2C peripheral?

Yes, but you cannot send gcode or Grbl style jog commands - only single character commands as [listed here](https://github.com/grblHAL/Plugin_keypad#keypad-plugin).

AvatarBg111, 2023-09-17:
> Which board map are you using? Asking since it could be due to pin assignment clashes. I do not lose USB connection when testing with my dev board.
> 
My board is DevEBox V3.0 which uses the microcontroller stm32f407vgt6. The USB OTG peripheral uses pins PA11 and PA12, while UART1 uses pins PA9 and PA10, UART2 uses pins PA2 and PA3, UART3 uses pins PD8 and PD9, and UART6 uses pins PC6 and PC7.

I looked in the code a couple of days ago and in the file driver.c at line 2486 there is this piece of code:
> #if USB_SERIAL_CDC
>     stream_connect(usbInit());
> #elif !defined(UART_INSTANCE)
>     stream_connect(serialInit(BAUD_RATE));
> #endif
In the function seralInit() UART1 is configured as a defualt serial port for receiving g-code by GRBL if the user does not want to use USB. When i "define" the macro MPG_ENABLE, somehow the firmware calls the function initializes UART1 as it's main serial port for receiving g-code and not USB. I found that out by hooking up a logic analyzer to the peripheral (UART1) and saw the initial report/message after startup of GRBL.

terjeio, 2023-09-17:
> When i "define" the macro MPG_ENABLE, somehow the firmware calls the function initializes UART1 as it's main serial port for receiving g-code and not USB.

Odd. If you introduce a bug in the code by changing `stream_connect(usbInit());` to `stream_connect(noInit());` does it still compile?

> I found that out by hooking up a logic analyzer to the peripheral (UART1) and saw the initial report/message after startup of GRBL.

Note that in MPG mode some of the output to the USB port is replicated to the MPG port to allow the MPG to display data without requesting it. E.g. the init message and the real-time report is.

AvatarBg111, 2023-09-18:
> Odd. If you introduce a bug in the code by changing `stream_connect(usbInit());` to `stream_connect(noInit());` does it still compile?
Since i didn't find a function noInit() i put a NULL as a parameter to stream_connect() (I also defined MPG_ENABLE with a value of 2). The firmware compiles. I flashed it into the MCU and when i opened my app for streaming g-code through USB (it's called LaserGRBL) from my PC, but it could not detect an available COM port. Which means that the problem is not that the USB connection cannot be made, but somehow the communication between the app on my PC and the MCU is not right. Maybe the  app does not connect to the MCU because it awaits some kind of acknowledge...
I'll try another g-code streaming app. Any recomendations? (My OS is Windows 10)

AvatarBg111, 2023-09-18:
> I'll try another g-code streaming app
>
I tried UGS (Universal G-code Sender) and the app connected...
I'm going to try to switch to MPG mode by sending a 0x8B to the MPG UART.

terjeio, 2023-09-18:
> Since i didn't find a function noInit() i put a NULL as a parameter to stream_connect() 

The idea was to introduce a bug, if it compiles then the USB port is not initialized as it should and the next step is to figure out why.

> I'll try another g-code streaming app. Any recomendations? (My OS is Windows 10)

[ioSender](https://github.com/terjeio/ioSender) is made to support most of grblHALs new features.