grblHAL/Plugin_keypad #8: Problem with plugins KEYPAD and DISPLAY

zab-ig, 2023-10-02:
Hello.
; grblHAL
; 1.1f.20230926
; [OPT:VNMSL,35,1024,3,0]
; [NEWOPT:ENUMS,RT+,ES,REBOOT,SED,RTC,ETH,ODO]
; [FIRMWARE:grblHAL]
; [NVS STORAGE:*FRAM 8K]
; [FREE MEMORY:492K]
; [DRIVER:iMXRT1062]
; [DRIVER VERSION:230926]
; [DRIVER OPTIONS:USB.2]
; [BOARD:GRBLHAL2000 - PRINTNC]
; [AUX IO:4,3,0,0]
; [IP:192.168.0.201]
; [NETCON:Telnet]
; [PLUGIN:MODBUS v0.15]
; [PLUGIN:MODVFD v0.03]
; [PLUGIN:KEYPAD v1.34]
; [SPINDLE:PWM]
; [PLUGIN:I2C Display v0.06]
; [PLUGIN:ODOMETERS v0.04]

There were problems activating the keypad and display plugins.
I use a keypad https://github.com/Expatria-Technologies/RT_Jog_Controller.
If you first connect power to the keypad and then apply power to the GRBL controller, then there are no problems with the operation of the keypad and the GRBL controller - everything works fine.
The screenshots show information from the i2c bus at the moment of switching on. You can see how the GRBL controller accesses the keypad and then the eeprom.

[image]
[image]

But if you connect the power to the keypad from the GRBL controller, i.e. power is supplied simultaneously. Or the keypad is not connected to the GRBL controller at all, then the GRBL controller does not work properly.
The screenshots show that the GRBL controller is trying to communicate with the keypad, but does not receive a response from it, after which the GRBL controller does not try to contact the EEPROM. There is no further communication on the i2c bus. If you send the $$ command, a list of parameters is displayed with default values, not the values ​​saved in eeprom.
[image]
[image]

zab-ig, 2023-10-02:
Thanks for the quick response.
10 ms was not enough. I tried 100, 150, 200 ms - it didn't work. As a result, I set the delay to 500 ms, with this delay the keypad and grbl controller work stably.
And another question: if you connect the keypad after turning on the Grbl controller, the keypad does not work and the Grbl controller does not respond to it. Can this be fixed somehow?

terjeio, 2023-10-02:
> Thanks for the quick response.

You were lucky as I was just about to commit a batch of other changes...

>  As a result, I set the delay to 500 ms, with this delay the keypad and grbl controller work stably.

That is a long time, does the keypad code really start up that slow?

> if you connect the keypad after turning on the Grbl controller, the keypad does not work and the Grbl controller does not respond to it. Can this be fixed somehow?

If the MCU I2C port tolerates hot inserts then either a $ command to initialize the plugin or an "inserted" signal must be added. Polling for it is, IMO, not a good idea.

terjeio, 2023-10-02:
I'll increase the delay to 510ms in the next commit since there is [500ms of delay](https://github.com/Expatria-Technologies/RT_Jog_Controller/blob/e10739204a813d3e47853ccb98ac789f7075fb14/FW/i2c_responder/app_main.cpp#L815-L817) in the startup sequence before I2C is enabled.

closed 2023-12-15 by terjeio