grblHAL/Plugin_keypad #10: Problem enabling macros

pimsierhuis, 2024-02-09:
I'm trying to enable macros but am puzzled by the following issue. I use picocnc board with rp2040 and use the web builder to create firmware. When I use "MACROS_ENABLE=3", the keypad (over serial) works, but $490 gives an error:53 (probably due to AUX input pin shortage). But when I use "MACROS_ENABLE=2" (the way I want to use it), the control-connection over USB stops working! I've looked at macros.c but can't find any explanation. 

Help would be very welcome!

terjeio, 2024-02-10:
Seems it is due to a memory leak - can you try again now?

pimsierhuis, 2024-02-10:
It works now. Great, thanks!

For my understanding: memory leak where? In the web builder?

terjeio, 2024-02-10:
In the macro plugin, bad code on my part. Will commit fix later.

pimsierhuis, 2024-02-11:
OK, check. Thanks for the quick fix!

closed 2024-02-16 by terjeio