grblHAL/Plugin_keypad #13: Keypad with LPC_176x

paukstelis, 2024-08-15:
I'm exploring the idea of using a keypad with LPC_176x (BTT SKR Turbo 1.4). Do I need to explicitly initialize TX0/RX0 (TFT connection) in order for it to accept the key commands?

terjeio, 2024-08-16:
> Looking through how it is implemented in the other drivers, I guess this was just never implemented for LPC176X?

Correct.

Add the keypad plugin to the build and change the [init function](https://github.com/grblHAL/Plugin_keypad/blob/5ea4bad8f08320f30dde397d95565e5082b42a96/keypad.c#L497) to open the port and register it:

```
bool keypad_init (void)
{
    if((nvs_address = nvs_alloc(sizeof(jog_settings_t)))) {

        hal.driver_cap.mpg_mode = stream_mpg_register(stream_open_instance(MPG_STREAM, 115200, NULL, NULL), false, stream_mpg_check_enable);

        ...
```

and add 

`#define KEYPAD_ENABLE            2`

and you should have keypad & MPG mode active, mode switching by the `0x8B` real time command character.

I'll add full MPG & keypad support later.

paukstelis, 2024-08-16:
Thanks for that!
It did make some progress. It does now recognize the 0x8B sent by a button push (at least, status report now has MPG: 1 and some other information in it immediately after pressing). I can't seem to get the axes to jog just yet.

terjeio, 2024-08-16:
Single character jog commands are only available when MPG mode is off (`MPG:0`), when on you are in full control and have to send normal jog commands.

paukstelis, 2024-08-17:
Still no joy there. I am sending 0x85 when jog buttons are released, followed by status report (`?`). I see that status report coming through the USB sender when the button is released, so it seems that is working. Just no movement in either MPG state.

paukstelis, 2024-08-17:
In my uninformed testing, it seems like all the real-time commands work, but `grbl.enqueue_gcode` doesn't seem to do anything. My testing is very naive, but I still don't get any jogging if I isolate the gcode command (to rule out the command building process) and try to get it to be enqueued:
```
            case 'H':                                   // Home axes
                //strcpy(command, "$H");
            	strcpy(command, "$J=G91 G21 X-500 F100");
            	jogging = grbl.enqueue_gcode((char *)command);
                break;
```
(I've changed remote button to send `H` in this case)
I tried a non-jog commands as well (G92 X), and that also doesn't seem to get enqueued.

terjeio, 2024-08-17:
I have just committed the needed changes to the LPC176x driver for proper keypad support.

paukstelis, 2024-08-17:
Tested it out and it works great. Thank you!

closed 2024-08-17 by paukstelis